ProjectsWithRed/auto-wire-stripper-cutter #4: Having issues loading the code. Can you help, please?

umalave, 2022-06-05:
Arduino: 1.8.19 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 115200, None"

AutoWireCutterStripper:60:71: error: stray '\302' in program

   <title>auto-wire-stripper-cutter/AutoWireCutterStripper.ino at main · ProjectsWithRed/auto-wire-stripper-cutter · GitHub</title>

                                                                       ^

AutoWireCutterStripper:60:72: error: stray '\267' in program

   <title>auto-wire-stripper-cutter/AutoWireCutterStripper.ino at main · ProjectsWithRed/auto-wire-stripper-cutter · GitHub</title>

                                                                        ^

AutoWireCutterStripper:60:116: error: stray '\302' in program

   <title>auto-wire-stripper-cutter/AutoWireCutterStripper.ino at main · ProjectsWithRed/auto-wire-stripper-cutter · GitHub</title>

                                                                                                                    ^

AutoWireCutterStripper:60:117: error: stray '\267' in program

   <title>auto-wire-stripper-cutter/AutoWireCutterStripper.ino at main · ProjectsWithRed/auto-wire-stripper-cutter · GitHub</title>

                                                                                                                     ^

C:\Users\myuser\Desktop\AutoWireCutterStripper\AutoWireCutterStripper.ino:411:10: warning: missing terminating ' character

     </option></form><form class="js-site-search-form" role="search" aria-label="Site" data-scope-type="Repository" data-scope-id="466200308" data-scoped-search-url="/ProjectsWithRed/auto-wire-stripper-cutter/search" data-owner-scoped-search-url="/users/ProjectsWithRed/search" data-unscoped-search-url="/search" data-turbo="false" action="/ProjectsWithRed/auto-wire-stripper-cutter/search" accept-charset="UTF-8" method="get">

          ^

AutoWireCutterStripper:411:10: error: missing terminating ' character

     </option></form><form class="js-site-search-form" role="search" aria-label="Site" data-scope-type="Repository" data-scope-id="466200308" data-scoped-search-url="/ProjectsWithRed/auto-wire-stripper-cutter/search" data-owner-scoped-search-url="/users/ProjectsWithRed/search" data-unscoped-search-url="/search" data-turbo="false" action="/ProjectsWithRed/auto-wire-stripper-cutter/search" accept-charset="UTF-8" method="get">

          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

AutoWireCutterStripper:468:75: error: stray '\342' in program

       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">↵</span>

                                                                           ^

AutoWireCutterStripper:468:76: error: stray '\206' in program

       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">↵</span>

                                                                            ^

AutoWireCutterStripper:468:77: error: stray '\265' in program

       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">↵</span>

                                                                             ^

AutoWireCutterStripper:473:56: error: stray '\342' in program

       <span class="d-inline-block ml-1 v-align-middle">↵</span>

                                                        ^

AutoWireCutterStripper:473:57: error: stray '\206' in program

       <span class="d-inline-block ml-1 v-align-middle">↵</span>

                                                         ^

AutoWireCutterStripper:473:58: error: stray '\265' in program

[... 712 lines cut ...]
AutoWireCutterStripper:2602:74: error: expected unqualified-id before '<' token

           <td id="LC285" class="blob-code blob-code-inner js-file-line">}</td>

                                                                          ^

AutoWireCutterStripper:2628:74: error: expected unqualified-id before '<' token

           <td id="LC291" class="blob-code blob-code-inner js-file-line">}</td>

                                                                          ^

AutoWireCutterStripper:2703:16: error: expected unqualified-id before numeric constant

         &copy; 2022 GitHub, Inc.

                ^~~~

exit status 1

stray '\302' in program

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

umalave, 2022-06-05:
I made a mistake when uploading the file to the ESP32. Problem fixed

steveyb109, 2022-06-21:
I cant get this to work all wired up and programmed but no screen and motor tries to move when i hold reset ? Please help esp32 vroom32

steveyb109, 2022-06-22:
Hopefully this will help, I've double checked all connections, there is 5v to the screen i think its a programming issue that could be me as I've not used Arduino before

steveyb109, 2022-06-22:
[image]

ProjectsWithRed, 2022-06-25:
Its not clear to see all the wiring, have you double checked all your wiring?

steveyb109, 2022-06-25:
I have 12v going into the step down and 5 volts coming out.
(Screen) 5v on VDO, GND to earth, SCK to G22, SDA to G21.
(Switch) 5v in on +, GND to earth, SW to G4, DT to G0, CLK to G23.
(Lateral movement) VMOT 12v in, GND to earth bridged with cap. VDD 5v in, GND to earth. MS1 5v and bridged to MS2, MS2 bridged to MS3, RESET BRIDGED to SLEEP, STEP to G18, DIRECTION to G5.
(Feed movement) VMOT 12v in, GND to earth bridged with cap. VDD 5v in, GND to earth. 5v to MS1 MS2 MS3, SLEEP and REST bridged, STEP to G16, REST to G17.
(Main Board) GND to earth, 5v to V5

steveyb109, 2022-06-28:
[image]

steveyb109, 2022-06-28:
All grounds go to the same place as above

ProjectsWithRed, 2022-07-01:
The wiring is so unclear that there could be so many things wrong that are hard to spot. Are you able to make a test version on a breadboard?

steveyb109, 2022-07-01:
Strip board and headers ect have arrived today going to start over as you suggest

steveyb109, 2022-07-01:
Plugged in the board on its own to the computer and it constantly reboots with the same errors as above

JoeKingSid, 2022-07-05:
I have the same issue!  Looking forward to seeing your solution!

steveyb109, 2022-07-05:
Im waiting for parts to arrive to finish my re-build, meanwhile I have used the ardinuo cloud tool to upload the sketch and the board no longer constantly re-boots. Might be worth a try if you have that same issue

steveyb109, 2022-07-07:
[image]

Board all re-wired but the main thing was use the cloud editor to upload the sketch. Also once you have calibrated wire length you need to change CALIBRATION_MODE = to false. So the unit will then cut multiple lengths.

ProjectsWithRed, 2022-07-08:
@steveyb109 So is everything working for you now?

steveyb109, 2022-07-08:
> 

Yes all working, thank you

ProjectsWithRed, 2022-07-08:
Perfect! No worries 👍 .