ProjectsWithRed/auto-wire-stripper-cutter #5: Arduino Programming issue

JoeKingSid, 2022-07-03:
Hello, I cannot seem to connect to my ESP board to upload the Sketch.  I think it is because I do not have the correct board selected.  I purchased this board [https://www.amazon.com/dp/B09J95SMG7?ref=ppx_yo2ov_dt_b_product_details&th=1](url) installed the required libraries but upon upload get an error message that states my board is not connected.  FYI... I have connected power, the screen and the pot on a breadboard just to keep things simple to start.  Can you advise what I need to do to get my Board to talk to my Computer?  Thanks!

JoeKingSid, 2022-07-04:
Here is a screen shot of the error message as well as my Board Config

[image]
g

JoeKingSid, 2022-07-04:
1 More shot of Device Manager recognizing the port 5 connection.  Thanks in advance for any advice/help!
[image]

ProjectsWithRed, 2022-07-04:
Try selecting "ESP32 Dev Module" as the board. Also press and hold the boot button on the ESP32 when you press upload.

JoeKingSid, 2022-07-04:
Well, actually I might have spoken a bit too soon ...  I found this thing in Arduino called Serial Monitor...  it seems to show that the Controller has some kind of error and is stuck in some kind of a Reboot Loop...  this is what it says...

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
E (121) gpio: esp_ipc_call_blocking failed (0x103)
E (121) gpio: gpio_install_isr_service(449): GPIO isr service already installed
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400ee48b  PS      : 0x00060533  A0      : 0x800d4d4f  A1      : 0x3ffe3b20  
A2      : 0x00000000  A3      : 0x3ffc1c38  A4      : 0x3ffc1c38  A5      : 0x00060523  
A6      : 0x00060520  A7      : 0x00000001  A8      : 0x800d490c  A9      : 0x3ffe3ae0  
A10     : 0x3ffbdca4  A11     : 0x00800000  A12     : 0x00000017  A13     : 0x3ffe3b20  
A14     : 0x007bdc90  A15     : 0x003fffff  SAR     : 0x00000009  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400865e5  LEND    : 0x400865f5  LCOUNT  : 0xfffffff7  

Backtrace:0x400ee488:0x3ffe3b200x400d4d4c:0x3ffe3b40 0x400d2cf4:0x3ffe3b70 0x400d2d75:0x3ffe3ba0 0x400d13b9:0x3ffe3bc0 0x400d15b9:0x3ffe3be0 0x400d96ab:0x3ffe3c10 0x40082ce1:0x3ffe3c40 0x4007920a:0x3ffe3c90  |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

It just keeps repeating that over and over.  Any ideas???  I wish you would make a video showing details of how to upload the INO cause so far everything else has been straightforward but, to a NOOB like me, the Arduino programming is not very intuitive.  THANKS!!!

ProjectsWithRed, 2022-07-08:
@JoeKingSid Someone had this similar problem, please see issue #4. To fix it, try uploading the code using the Arduino web IDE.

JoeKingSid, 2022-07-23:
UPDATE... First let me say thanks for the suggestions thus far!  To date I have still not been able to get the project working. :-( I was able to upload code via the Online Web Editor and I think that part is complete but honestly I'm not sure. After uploading the code I was expecting to see my OLED display light up with something but, No Joy.  I decided to start from scratch and these are the steps I took...
1. Connect the ESP32 to the Computer via USB cable.
2. Upload Code
3. Place ESP32 chip on Breadboard and connect  12vdc adapter to Stepdown Converter
4. verify 5vdc on stepdown and connect  5vdc wires as shown on diagram to ESP32 & OLED display
5. Connect OLED SDK & SDA  to ESP Pins 21 & 22 as shown
6. Connect Pot/Encoder to 5VDC (as shown) & SW, DT & SDK to ESP32 Pins  as shown.

applied power to circuit - nothing
pressed enable button on ESP32 - nothing
pressed boot button on ESP32 -nothing

Tried Re-uploading the code - nothing

Not really sure what else to try as this seems to be the "bare bones" setup in which something should come to life.

I've done lots of programming on Industrial PLC's and also have worked with the Arduino MEGA 2560 series of boards/ uploaded code successfully and don't understand what I am doing wrong that is causing me so many issues.  

I really do like the idea of this project and would like to realize success but I'm thinking that perhaps the ESP32 controller is not the best way to go for me?  Was going to look at maybe trying to convert this over to some other Native Arduino Board format and giving it a go that way?  

Any advice is certainly appreciated!  Thanks in advance and I hope somebody somewhere has an idea of how to get things going!

ProjectsWithRed, 2022-07-25:
Hello Joe, thank you for all this information. I would suggest completing all of the wiring on the breadboard, then turning everything on. I have not tested the code with partial wiring of the components. I think that is what is causing you issues. So complete the full wiring, send me a picture of it if you can and we will go from there.

ProjectsWithRed, 2022-07-25:
@JoeKingSid Also, just to make sure, did you change the i2c address of your OLED display in the code? Make sure `#define i2c_Address 0x3c` corresponds to the address of your OLED display which you will find out from the datasheet of the OLED display. If you did not change this, try again without the full wiring.

JoeKingSid, 2022-07-25:
OK, Thanks I will try those items and report back!

JoeKingSid, 2022-07-31:
OK!  Success!!! 

BTW - It is possible to test the display without all the wiring being connected.  I was able to achieve this by using the 3.3v output from the ESP32 (instead of the 12vdc/5vdc system that is shown on your schematic. After verifying that the display/pot were working in this config, I switched back to the shown design and found loose connections in the 5VDC distribution and analog outputs which were causing my problems.

Also, I was only able to upload the INO if I hold down the ENABLE Button on the ESP32 while it is compiling/Uploading which I feel should definitely be mentioned.

Thanks much for your help/patience! Now, let's get those wires STRIPPED!!!

ProjectsWithRed, 2022-08-01:
This is great to hear! I am glad everything worked out 👍🏼.

closed 2022-08-01 by ProjectsWithRed