ProjectsWithRed/auto-wire-stripper-cutter #7: Fusion Repo?

jrsteensen, 2022-08-04:
Hello! I was just curious if the Fusion repository was  publicly available? I'm going to build this but wanted to make a few changes in MCAD for personal preferences. Fantastic project and really excited to work on it.

Thanks!

ProjectsWithRed, 2022-08-24:
Unfortunately I can't provide the Fusion 360 file, however, feel free to import the model to F360 and modify it with the editing tools to meet your needs 👍🏼.

leahcornelius, 2023-04-14:
Hi, I was wondering if it is specifically F360 files you can't provide or general-cad files. Would you be able to export a .step file or similar, or is it that you cannot upload the CAD designs themselves (completely understandable, I've had projects myself I wasn't allowed to publish the cad for).

Really awesome work though! Just a few changes and tweaks I want to add - I can try and remake the CAD bodies using the 3mf files but if you have them to hand already would be silly to do so.

ProjectsWithRed, 2023-04-16:
@leahcornelius The stl files are also available in the [stl directory](https://github.com/ProjectsWithRed/auto-wire-stripper-cutter/tree/main/3d_printing/stl), are you not able to use those?

ProjectsWithRed, 2023-04-16:
@leahcornelius The stl files are also available in the [stl directory](https://github.com/ProjectsWithRed/auto-wire-stripper-cutter/tree/main/3d_printing/stl), are you able to use those?

leahcornelius, 2023-04-16:
Sadly not, as they are meshes and so cannot be edited in cad software simply. I could convert them into bodys/cad files manually but will probably just produce my own designs for now as it will take the same amount of time. Awesome work though! Its a really well thoight out design and will be a solid inspiration for projects

ProjectsWithRed, 2023-04-16:
@leahcornelius Will `.step` files help you? Let me know what extension is best for you. Thanks! I appreciate that.

leahcornelius, 2023-04-16:
@ProjectsWithRed yeah step files would be great! I use fusion 360 so .f3d, .step, all work. Thanks

ProjectsWithRed, 2023-04-16:
@leahcornelius You can now find the `.step` file for the full design in `3d_printing/assembly_model.step` 👍🏼.

leahcornelius, 2023-04-16:
Thanks, i appricate it!

closed 2023-04-16 by ProjectsWithRed