ProjectsWithRed/auto-wire-stripper-cutter #8: Serial monitor problems

dhardyuk, 2022-09-01:
I usually stick some variables in the start of the code with the sketch name and version etc. This then gets output to the serial monitor and/or displayed on the screen as part of the setup() code.

So I've copied over your code and added this to the variables section at the top after the #includes:

//Name and version
const String progname = "Wire Cut and Strip";
const String progver = "0.01";

And this to the start of the setup() code:

// push out some details over the serial port
//    delay(2000);
//    Serial.begin(9600);//Set up serial stream
//    delay(2000);
//    Serial.print("Programe Name: ");
//    Serial.print (progname);
//    Serial.print ("\n");
//    Serial.print("Version:       ");
//    Serial.print (progver);
//    Serial.print ("\n");
//    delay(2000);
//    Serial.end();//end serial stream  

So far so good. However, when I remove the // for the Serial output stuff the serial monitor just loops through this text:

20:11:33.156 -> Programe Name: Wire Cut and Strip
20:11:33.200 -> Version:       0.01
20:11:39.169 -> Programe Name: Wire Cut and Strip
20:11:39.169 -> Version:       0.01
20:11:45.178 -> Programe Name: Wire Cut and Strip
20:11:45.178 -> Version:       0.01
20:11:51.156 -> Programe Name: Wire Cut and Strip
20:11:51.202 -> Version:       0.01
20:11:57.160 -> Programe Name: Wire Cut and Strip
20:11:57.205 -> Version:       0.01

This is lifted straight from my other sketches, with the variables set to different values. Any idea why this is happening or where I should start trying to debug?

I don't have anything else connected, no buttons or rotary encoder 

I have adjusted your code slightly to make if fit an Arduino Nano v3.0 on a 3 Stepper CNC shield HW-702 - the tweaks needed are to set the pins for the A4988 stepper drivers to suit the board (so using Y axis stepper pins 6 and 3 for extruder and Z axis stepper pins 7 and 4) The temporary setup buttons are on 5 and 8, the encoder DT and clock are 12 and 13, and the encoder button is 9. However, ignoring all that and using your code without those tweaks gets me to the correct display.

dhardyuk, 2022-09-01:
Update - rotary encoders delivered so I have that part installed and working.

dhardyuk, 2022-09-02:
To add to the above, the looping happens and the screen never gets drawn. Its power cycling after the setup() code. Comment out the serial code and it runs find.

ProjectsWithRed, 2022-09-02:
Hello! It looks like it's rebooting everytime which is causing the "loop" of the setup function. Below are some things that may help you debug the problem.

- You might have connected something to the reset pins of the micro-controller, or to pins that should not be used, so do double check the datasheet to ensure you are not using pins that you shouldn't be using.
- Try keeping all Serial statements uncommented apart from `Serial.end()`.
- Start a new blank sketch, with nothing connected to the micro-controller, add your code, run it to ensure all good, then slowly, bit by bit add components and code to see exactly what is causing the problem.

Let me know how it goes👍.

dhardyuk, 2022-09-02:
It fails the same way. It has to be something weird!

ProjectsWithRed, 2022-09-02:
Yes it is! Did you manage to do point 3 that I mentioned? At what point did it fail?

One quick thing to try is, remove the string variables that you have, `progname` and `progver`, and where you have referenced them just use a hard-coded string value.

Side note: Rather than doing `Serial.print ("\n");`, just use `Serial.println()` in the previous statement, it will add a line break at the end for you.

ProjectsWithRed, 2022-09-28:
@dhardyuk Was this issue fixed?