ProjectsWithRed/auto-wire-stripper-cutter #9: Motor only turns once and stops

GuilhermeLeonidio, 2022-09-16:
hello, how are you? I'm using the green A9488 module instead of the red one and I'm trying to make only one engine work but it spins for about 5 seconds and doesn't do anything else, I didn't add any more lines in the code and I didn't put any numbers or letters plus this is 100% original. I would like to know if there is something that needs to be put in the code or needs the three engines to work, I would be grateful if you answer me this would help me a lot my friend
[image]

ProjectsWithRed, 2022-09-16:
Yes you are supposed to make some changes to the code, and the changes are commented directly in the code, so I recommend going through the code and reading the comments to see exactly what changes you need to make and how to use it.

By default, calibration mode will be on (see the `CALIBRATION_MODE` variable). 

Are you referring to the stepper motor as the "engine"? If so, does it turn when you turn on power, or after you press the rotatary encoder?

GuilhermeLeonidio, 2022-09-16:
Hello all well? it spins after pressing the rotary encoder, I'm going through and reading the comments but I'm quite a layman in this part of c++ communication could you give me some help in this part of the adjustment? a picture of something like that with the modifications

GuilhermeLeonidio, 2022-09-16:
I need to set the calibration_MODR to false ?

AV1987, 2022-09-24:
How did you get the Oled display turned on? Mine doesn't display.

ProjectsWithRed, 2022-09-28:
@GuilhermeLeonidio `CALIBRATION_MODE` is used to adjust `WIRE_MOVEMENT_MULTI` to make sure the extruder stepper motor (the one that moves the wire) gives the correct wire length. So if you have `CALIBRATION_MODE` to true, you need to put a number on the first cell (top left, only that cell will work in calibration mode). So now when you press the encoder, the wire will feed and then the blades will cut the wire without stripping, then you need to measure the wire length, and see if it is longer or shorter than the measurement you put in the first cell, and from there adjust `WIRE_MOVEMENT_MULTI` to be higher or lower depending if the wire was longer or shorter than the measurement you put in the first cell.

After you have achieved the correct wire sizes, you can then make `CALIBRATION_MODE` to false, since you have now finished the calibration.

If you want to run the machine without calibration, just make `CALIBRATION_MODE` to false, however, the wire lengths might not be accurate, since why you need to calibrate.

Let me know if you have any questions.

closed 2023-04-16 by ProjectsWithRed