ProjectsWithRed/auto-wire-stripper-cutter #10: first wirring: the 3 stepper motor turn slowly

Parang0n, 2022-09-21:
hello and firstly , thanks for you DIY! this is a masterpiece that i NEED to see working ( on my house °_° ) , sorry for my english ( hello from france )

i have make all the wirring and when i start the wire cutter , ( 1st the screen is black ) , but most important the 3 motor turn very slowly and seems to dont want stop 

lm2596 IN : 12V  OUT: 5.08V

i have make the solution to press boot when a upload the file and i use "esp 32 dev module "

shall you have the time to help me : i real want to make this diy , thanks a lot

ProjectsWithRed, 2022-09-28:
@Parang0n Hello, thank you! 

I have made an update to the code, please upload again and let me know if you still have issues.

jagallja, 2022-10-22:
i have the same problem and can not figure the solution, and when i connect the z steppers the screen goes black

jagallja, 2022-10-22:
Hi @ProjectsWithRed 
seem like the updated code still have issues in it, my 3 steppers still spinning on it own :(

ProjectsWithRed, 2022-10-27:
@jagallja Have you adjusted the bolt on the A4988 drivers to match your stepper motor drivers? Check this [link](https://lastminuteengineers.com/a4988-stepper-motor-driver-arduino-tutorial/) for the full explanation and how to adjust your drivers.

jagallja, 2022-11-02:
@ProjectsWithRed accidentaly solved this issue by lifting my PCB, it might short circuit or grounding. thank you for the awesome projects 👍

and i have one more problem, how do i reverse the direction of my extruder ? flipping the wire like 3D printer extruder doesnt work in it

Parang0n, 2022-11-03:
Hello sorry for the time , but i need to give you some information: 

1st : the wire cutter WORK !!!! thank a lot it's a masterpiece.

2nd : it have a little bug but its not bad (70% of the time)  when you start the cutter, the 2 up/down motor make noise like they blocked , i think it come to the fact that the 2 motor have not the same position with wiring A and B  and the A4988 dont know what to do. don't worry i use the project like that

there is a picture of your invention ! 

thanks a lot !!!

[20221103_235406](https://user-images.githubusercontent.com/113788664/199850502-8984b071-835c-47af-8d34-2ef3b6bca27a.jpg)

ProjectsWithRed, 2022-11-08:
@Parang0n Perfect glad to hear its working now! To reverse the direction you can use negative steps. One way you can do it is change the `WIRE_MOVEMENT_MULTI` variable to be a negative value, for example, if it was 400, make it -400.

ProjectsWithRed, 2022-11-08:
@jagallja Perfect glad to hear its working now! To reverse the direction you can use negative steps. One way you can do it is change the WIRE_MOVEMENT_MULTI variable to be a negative value, for example, if it was 400, make it -400.

ProjectsWithRed, 2022-11-08:
@Parang0n That's awesome!! Thanks for sharing the picture, I love it!

The blade stepper motors might experience "blocking" as you said when they are not perfectly level when starting, so just make sure you calibrate them before turning on the machine. You can also look into changing the current for the stepper motors by adjusting the stepper motor drivers screw, see this [link](https://lastminuteengineers.com/a4988-stepper-motor-driver-arduino-tutorial/) which shows how to do that, ensure you are supplying the stepper motors sufficient current.

closed 2023-04-16 by ProjectsWithRed