ProjectsWithRed/auto-wire-stripper-cutter #11: No display

AV1987, 2022-09-22:
Hi @ProjectsWithRed.
Thanks for sharing this project, very interesting.
It seems I uploaded the code sucessful but I have no display. I only connect ESP32 Devkit with Oled 128x64.
The sample code works ok with display. What do I miss?
Thank you

Tips: adding a 10uF Cap on RTS pin to ground to avoid holding the Reset button during sketching.

ProjectsWithRed, 2022-09-28:
I have updated the code since there was an update to the `Encoder` library which made it not work. Upload again and let me know if you still have issues.

AV1987, 2022-09-28:
@ProjectsWithRed
Thank you. My display works now. But none of the motor runs and pulling 1.7A. It's hot the driver. Do you have this problem? I checked the wiring 2B,2A,1B,1A are correct and Calibration is false by default of your updated sketch.

AV1987, 2022-09-29:
> @AV1987 I have updated the code since there was an update to the `Encoder` library which made it not work. Upload again and let me know if you still have issues.

Thanks Red! My setup is running now. Will give you update.

AV1987, 2022-09-29:
Hi @ProjectsWithRed

I have everything is running now with your original updated code. However, the linear motion is running very slow and dragging not like your youtube video.
Please advise. Thanks

closed 2022-09-30 by ProjectsWithRed