ProjectsWithRed/auto-wire-stripper-cutter #12: Motor speed in slow motion

AV1987, 2022-09-29:
Hi @ProjectsWithRed
I got my setup is running, I leave everything same as your original sketch it seems matches with my calibration.
A new problem for me why these motors are moving very slow like "slow motion"
Please advise
Thanks

ProjectsWithRed, 2022-09-29:
Hello @AV1987, that's great to hear! Have you adjusted the A4988 stepper driver by turning the little bolt on it? That controlls how much current flows to the stepper motor, which may be causing the slow movement. Check this link which explains it well and how to adjust for your stepper motor.

AV1987, 2022-09-29:
Hi @ProjectsWithRed
I did try both increase current and voltage on 3x A4988 drivers. All of them slow like slow motion :)
Also, each of them pulling only 250mA
Whatelse could it be?

ProjectsWithRed, 2022-09-29:
What step configuration do you have on the A4988 drivers (the MS1, MS2, and MS3 pins)? You can also try to increase the speed of the motor by using these two variables, `LINMOT_STEPPERS_SPEED ` and `EXTRUDER_STEPPER_SPEED `.

ProjectsWithRed, 2022-09-30:
Can you link the exact motors used? Also, try increasing the `LINMOT_STEPPERS_SPEED ` and `EXTRUDER_STEPPER_SPEED ` variables.

AV1987, 2022-09-30:
@ProjectsWithRed 
Good idea!
RTELLIGENT Nema 17 Stepper Motor 3PCS, 2 Phase Step Motor Bipolar 1.5A 59.5oz.in(42Ncm) 42x42x38mm 4-Wire 30cm Long Cable for 3D Printer (3, 42A02C-XH2.54) https://a.co/d/7eo0juY

AV1987, 2022-09-30:
@ProjectsWithRed 
Forgot to quote your replied: yes i did change it to 200, that didn't help. Here's exactly problem when I turn it on (after uploaded sketch):
1. Humming noise from the motors when turn on, they're pulling 600mA. 
2. No display until I press EN button on ESP32.
3. Press EN, the noise on motors is gone, no load. Display shows up. But slow motion still there when I run it.
Is it suppose to be normal?

`Can you link the exact motors used? Also, try increasing the LINMOT_STEPPERS_SPEED and EXTRUDER_STEPPER_SPEED variables.`

AV1987, 2022-09-30:
@ProjectsWithRed
!(https://user-images.githubusercontent.com/114119127/193358579-d41aae2e-6a1a-4ad1-a58e-ecf7f3f19b14.png)

ProjectsWithRed, 2022-10-27:
@AV1987 No its not normal. Are you sure you changed the bolt on the A4988 stepper drivers to the correct current for the motors? Check this [link](https://lastminuteengineers.com/a4988-stepper-motor-driver-arduino-tutorial/), which explains it in great detail.

closed 2023-04-16 by ProjectsWithRed