ProjectsWithRed/auto-wire-stripper-cutter #13: Step motor not turn  please help ...

JamerYou, 2023-02-11:
Hello sir i like to build this machine  but i got stuck here  the motor it not turn  even one step  , I check all connections everything exactly your picture diagram  then upload code but step motor not move  , display turn on good encoder work , but  all 3 step motor not even more , please help . Thank you .

ProjectsWithRed, 2023-03-22:
There could be a number of ways why this is the case and its hard to tell without knowing more information. However, things that come to mind are:
- To double check the wiring for the stepper motors.
- Ensure the stepper motor drivers you are using are rated for the stepper motors.
- Ensure you have adjusted the current limiter on each stepper motor driver to match the rated current of the stepper motors.
- Ensure that the enable pins are connected for each of the motor drivers, otherwise they will not work (see image below, black arrows).
 
If none of these points help, please share as much information as you can.

[image]

cwbullet, 2025-06-22:
I am having the same problem.  Is this project dead?