ProjectsWithRed/auto-wire-stripper-cutter #14: software installation problem

Ackerley35, 2023-03-15:
I am getting an error like this. Can you help me?

Leaving...
Hard resetting via RTS pin...

Arduino IDE 
Version: 2.0.4
CLI Version: 0.31.0

belenhed, 2023-03-21:
Hello maybe this can help?
https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/

ProjectsWithRed, 2023-03-22:
@Ackerley35 Are you holding down the boot button on the ESP32 when uploading the code? Also double check that you are using the correct board in the board manager in the Arduino IDE.

Ackerley35, 2023-03-22:
@ProjectsWithRed Despite that error, the program was loaded and this error did not matter. After checking their connections, the screen and stepper motors worked. But after a few tries, I think my esp32 module burned out :( When I try to upload it again by pressing the Boot button, I get an error "Failed uploading: uploading error: exit status 2". I think my card is probably broken and I'm planning to make software with arduino nano for now.
As if the ESP32 module is not very suitable. What do you think about this?

Also your idea is amazing and very simple. I congratulate you. Also sorry for my English, I have to answer with google translate.

belenhed, 2023-03-22:
Try to solder the cap 10 uF electrolytic capacitor between the EN pin and GND.

ProjectsWithRed, 2023-03-23:
> @ProjectsWithRed Despite that error, the program was loaded and this error did not matter. After checking their connections, the screen and stepper motors worked. But after a few tries, I think my [...]

That error doesn't necessarily mean the ESP32 is burned out. One way to check if it is, connect the ESP32 to power and see if the chip gets hot very quickly that you can't put your finger on it more than a couple of seconds, if that's the case then it might be burned out. If needed, share your wiring so we can see if there are any connections that might be causing the ESP32 to stop working.

No worries, I am glad to hear it!