ProjectsWithRed/auto-wire-stripper-cutter #15: No display on turn on with DT pin on encoder

belenhed, 2023-03-20:
Hello

I made this and when I turn it on I got no display. 
If I disconnect the dt pin of the encoder, and turn it on again I got display.
I manage to solve it putting and on off switch at the DT pin.
Turn it on whith the switch in off position and then when I got display turn the switch to on position for the encoder to work.
Do you have any ideas why I got this behavior?
Thanks

ProjectsWithRed, 2023-03-22:
I think this possibly has to do with the encoder library version used, or some small issue with wiring. Something to try is removing the line below in the `AutoWireCutterStripper.ino` file and see if the problem still persists.

https://github.com/ProjectsWithRed/auto-wire-stripper-cutter/blob/515eb32fa96801f713e04d3cdbdf00c595d80443/AutoWireCutterStripper/AutoWireCutterStripper.ino#L1

belenhed, 2023-04-02:
Hi I commented out the line but thew problem persist also I'm having issues sometimes it turns on and some times doesn't weird
I have to dc the esp from the headers and is back online